UshakovVasilii/gnome-shell-extension-freon #131: TypeError: ExtensionUtils.initTranslations is not a function

terrycloth, 2019-04-04:
I maintain the RPM package of Freon for Fedora. After updating to Freon version 36 on Fedora 29 Workstation with GNOME Shell 3.30.2, Freon no longer loads. GNOME Tweaks shows the error icon next to Freon, and when you click on the gear button, you get the error message below.

Possibly related to #126?
```
TypeError: ExtensionUtils.initTranslations is not a function

Stack trace:
  init@/usr/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/prefs.js:17:5
  _getExtensionPrefsModule@resource:///org/gnome/shell/extensionPrefs/main.js:68:9
  wrapper@resource:///org/gnome/gjs/modules/_legacy.js:82:22
  _selectExtension@resource:///org/gnome/shell/extensionPrefs/main.js:82:31
  wrapper@resource:///org/gnome/gjs/modules/_legacy.js:82:22
  _onCommandLine@resource:///org/gnome/shell/extensionPrefs/main.js:235:17
  wrapper@resource:///org/gnome/gjs/modules/_legacy.js:82:22
  main@resource:///org/gnome/shell/extensionPrefs/main.js:389:5
  @<main>:1:43
```

jlmagee, 2019-04-06:
I'm having this issue on F29 as well. Downgrading Freon to 34 does not solve the problem. It worked before these updates were installed this morning
Installing:
 kernel                                      x86_64 5.0.5-200.fc29                     updates                        30 k
 kernel-core                                 x86_64 5.0.5-200.fc29                     updates                        26 M
 kernel-devel                                x86_64 5.0.5-200.fc29                     updates                        13 M
 kernel-modules                              x86_64 5.0.5-200.fc29                     updates                        28 M
 kernel-modules-extra                        x86_64 5.0.5-200.fc29                     updates                       2.1 M
Upgrading:
 cockpit                                     x86_64 191-1.fc29                         group_cockpit-cockpit-preview  32 k
 cockpit-bridge                              x86_64 191-1.fc29                         group_cockpit-cockpit-preview 548 k
 cockpit-dashboard                           noarch 191-1.fc29                         group_cockpit-cockpit-preview 145 k
 cockpit-debuginfo                           x86_64 191-1.fc29                         group_cockpit-cockpit-preview  15 M
 cockpit-debugsource                         x86_64 191-1.fc29                         group_cockpit-cockpit-preview 297 k
 cockpit-doc                                 noarch 191-1.fc29                         group_cockpit-cockpit-preview 635 k
 cockpit-docker                              x86_64 191-1.fc29                         group_cockpit-cockpit-preview 372 k
 cockpit-kdump                               noarch 191-1.fc29                         group_cockpit-cockpit-preview 123 k
 cockpit-kubernetes                          x86_64 191-1.fc29                         group_cockpit-cockpit-preview 4.4 M
 cockpit-machines                            noarch 191-1.fc29                         group_cockpit-cockpit-preview 456 k
 cockpit-machines-ovirt                      noarch 191-1.fc29                         group_cockpit-cockpit-preview 454 k
 cockpit-networkmanager                      noarch 191-1.fc29                         group_cockpit-cockpit-preview 235 k
 cockpit-packagekit                          noarch 191-1.fc29                         group_cockpit-cockpit-preview 326 k
 cockpit-pcp                                 x86_64 191-1.fc29                         group_cockpit-cockpit-preview  74 k
 cockpit-selinux                             noarch 191-1.fc29                         group_cockpit-cockpit-preview 167 k
 cockpit-sosreport                           noarch 191-1.fc29                         group_cockpit-cockpit-preview  52 k
 cockpit-storaged                            noarch 191-1.fc29                         group_cockpit-cockpit-preview 394 k
 cockpit-system                              noarch 191-1.fc29                         group_cockpit-cockpit-preview 1.1 M
 cockpit-tests                               x86_64 191-1.fc29                         group_cockpit-cockpit-preview 262 k
 cockpit-ws                                  x86_64 191-1.fc29                         group_cockpit-cockpit-preview 788 k
 appstream-data                              noarch 29-10.fc29                         updates                        17 M
 audit                                       x86_64 3.0-0.7.20190326git03e7489.fc29    updates                       227 k
 audit-libs                                  x86_64 3.0-0.7.20190326git03e7489.fc29    updates                       106 k
 container-selinux                           noarch 2:2.94-1.git1e99f1d.fc29           updates                        46 k
 environment-modules                         x86_64 4.2.3-1.fc29                       updates                       309 k
 firefox                                     x86_64 66.0.2-1.fc29                      updates                        89 M
 flatpak                                     x86_64 1.2.4-1.fc29                       updates                       1.0 M
 flatpak-libs                                x86_64 1.2.4-1.fc29                       updates                       344 k
 fuse-overlayfs                              x86_64 0.3-8.dev.gita6958ce.fc29          updates                        49 k
 git-cola                                    noarch 3.3-1.fc29                         updates                       982 k
 gnome-shell-extension-no-topleft-hot-corner noarch 19.0-1.fc29                        updates                        17 k
 hwdata                                      noarch 0.322-1.fc29                       updates                       1.5 M
 java-1.8.0-openjdk                          x86_64 1:1.8.0.201.b09-6.fc29             updates                       239 k
 java-1.8.0-openjdk-devel                    x86_64 1:1.8.0.201.b09-6.fc29             updates                       9.8 M
 java-1.8.0-openjdk-headless                 x86_64 1:1.8.0.201.b09-6.fc29             updates                        32 M
 java-11-openjdk                             x86_64 1:11.0.2.7-7.fc29                  updates                       219 k
 java-11-openjdk-demo                        x86_64 1:11.0.2.7-7.fc29                  updates                       4.3 M
 java-11-openjdk-demo-slowdebug              x86_64 1:11.0.2.7-7.fc29                  updates                       4.3 M
[... cut ...]
 kernel-core                                 x86_64 4.20.14-200.fc29                   @updates                       61 M
 kernel-devel                                x86_64 4.20.14-200.fc29                   @updates                       51 M
 kernel-modules                              x86_64 4.20.14-200.fc29                   @updates                       28 M
 kernel-modules-extra                        x86_64 4.20.14-200.fc29                   @updates                      2.1 M

I couldn't spot the offender

tomporter518, 2019-04-08:
@jlmagee As I recall (when I ran into this), version 34 won't work for different reasons. I manually downloaded gnome-shell-extension-freon-35-1.fc29.noarch.rpm from koji and installed (over/downgraded via rpm) to get a working version. Until 36 is fixed or pulled from the repo, you will have to ignore it's upgrade request when checking. I add '--exclude=*freon' when running dnf.

UshakovVasilii, 2019-04-08:
`ExtensionUtils.initTranslations` exist in gnome-shell 3.32. Correct version for 3.30 should be provided by https://extensions.gnome.org/extension/841/freon/

closed 2019-04-08 by UshakovVasilii

jlmagee, 2019-04-08:
This is totally unresponsive for downstream Fedora 29 users. For at least the last 10 releases of Fedora, extensions.gnome.org produces this error

"We cannot detect a running copy of GNOME on this system, so some parts of the interface may be disabled. See our troubleshooting entry for more information."

 I'll try the @tomporter518 suggestion.

fyi @terrycloth

jlmagee, 2019-04-08:
@tomporter518 @terrycloth Thanks. Found a 35.1 rpm and it worked.  
@UshakovVasilii No thanks to you.

UshakovVasilii, 2019-04-09:
@jlmagee This problem related to rpm package, extension work on 3.30 and 3.32 fine.
master support 3.32 only.

UshakovVasilii, 2019-04-09:
[image]

terrycloth, 2019-04-09:
Ushakov is right, it was my mistake pushing Freon 36 to Fedora 29 in the first place. I've [pushed an update](https://bodhi.fedoraproject.org/updates/gnome-shell-extension-freon-35-3.fc29) to revert it back to Freon 35. It's in the "testing" phase right now, but I've confirmed it actually works this time, so in the meanwhile you can also [download the RPM](https://kojipkgs.fedoraproject.org//packages/gnome-shell-extension-freon/35/3.fc29/noarch/gnome-shell-extension-freon-35-3.fc29.noarch.rpm) to install right away.