UshakovVasilii/gnome-shell-extension-freon #134: Syntax Error: invalid regular expression flag 0

terrycloth, 2019-05-23:
I package Freon as an RPM for Fedora. I'm currently on Fedora 30 Workstation, with GNOME 3.32.2.

When updating my package to use your `EGO-37` git tag, I run into a new error. Looking Glass > Extensions > Freon > Show Errors says

> `Syntax Error: invalid regular expression flag 0`

I checked the systemd journal logs for GNOME Shell, but I'm not sure much the rest of this is relevant to Freon...

```
May 23 12:36:12 mars.localdomain gnome-shell[3124]: Extension "freon@UshakovVasilii_Github.yahoo.com" had error: SyntaxError: invalid regular expr>
May 23 12:36:57 mars.localdomain org.gnome.Shell.desktop[3124]: Window manager warning: Buggy client sent a _NET_ACTIVE_WINDOW message with a time>
May 23 12:36:58 mars.localdomain org.gnome.Shell.desktop[3124]: Window manager warning: Buggy client sent a _NET_ACTIVE_WINDOW message with a time>
May 23 12:38:17 mars.localdomain org.gnome.Shell.desktop[3124]: Window manager warning: META_CURRENT_TIME used to choose focus window; focus windo>
May 23 12:38:18 mars.localdomain org.gnome.Shell.desktop[3124]: current session already has an ibus-daemon.
May 23 12:38:18 mars.localdomain gnome-shell[3124]: g_dir_open_with_errno: assertion 'path != NULL' failed
May 23 12:38:18 mars.localdomain gnome-shell[3124]: g_filename_to_utf8: assertion 'opsysstring != NULL' failed
May 23 12:38:18 mars.localdomain gnome-shell[3124]: g_dir_open_with_errno: assertion 'path != NULL' failed
May 23 12:38:18 mars.localdomain gnome-shell[3124]: g_filename_to_utf8: assertion 'opsysstring != NULL' failed
May 23 12:38:18 mars.localdomain gnome-shell[3124]: Extension "freon@UshakovVasilii_Github.yahoo.com" had error: SyntaxError: invalid regular expr>
May 23 12:38:18 mars.localdomain gnome-shell[3124]: Some code accessed the property 'PREFS_SCHEMA' on the module 'prefs'. That property was define>
May 23 12:38:18 mars.localdomain gnome-shell[3124]: WARNING: addSignalMethods is replacing existing [object prototype of GType:Gjs_PomodoroTextInd>
May 23 12:38:18 mars.localdomain gnome-shell[3124]: WARNING: addSignalMethods is replacing existing [object prototype of GType:Gjs_PomodoroTextInd>
May 23 12:38:18 mars.localdomain gnome-shell[3124]: WARNING: addSignalMethods is replacing existing [object prototype of GType:Gjs_PomodoroTextInd>
May 23 12:38:18 mars.localdomain gnome-shell[3124]: WARNING: addSignalMethods is replacing existing [object prototype of GType:Gjs_PomodoroIconInd>
May 23 12:38:18 mars.localdomain gnome-shell[3124]: WARNING: addSignalMethods is replacing existing [object prototype of GType:Gjs_PomodoroIconInd>
May 23 12:38:18 mars.localdomain gnome-shell[3124]: WARNING: addSignalMethods is replacing existing [object prototype of GType:Gjs_PomodoroIconInd>
May 23 12:38:18 mars.localdomain gnome-shell[3124]: Some code accessed the property 'GPasteStatusIcon' on the module 'statusIcon'. That property w>
May 23 12:38:18 mars.localdomain gnome-shell[3124]: Some code accessed the property 'GPasteDummyHistoryItem' on the module 'dummyHistoryItem'. Tha>
May 23 12:38:18 mars.localdomain gnome-shell[3124]: Some code accessed the property 'GPasteSearchItem' on the module 'searchItem'. That property w>
May 23 12:38:18 mars.localdomain gnome-shell[3124]: Some code accessed the property 'GPastePageSwitcher' on the module 'pageSwitcher'. That proper>
May 23 12:38:19 mars.localdomain gnome-shell[3124]: Some code accessed the property 'GPasteUiItem' on the module 'uiItem'. That property was defin>
May 23 12:38:19 mars.localdomain gnome-shell[3124]: Some code accessed the property 'GPasteEmptyHistoryItem' on the module 'emptyHistoryItem'. Tha>
May 23 12:38:19 mars.localdomain gnome-shell[3124]: Some code accessed the property 'GPasteAboutItem' on the module 'aboutItem'. That property was>
May 23 12:38:19 mars.localdomain gnome-shell[3124]: Some code accessed the property 'GPasteStateSwitch' on the module 'stateSwitch'. That property>
May 23 12:38:19 mars.localdomain gnome-shell[3124]: Some code accessed the property 'GPasteItem' on the module 'item'. That property was defined w>
May 23 12:38:19 mars.localdomain gnome-shell[3124]: Some code accessed the property 'GPasteDeleteItemPart' on the module 'deleteItemPart'. That pr>
May 23 12:38:19 mars.localdomain gnome-shell[3124]: Some code accessed the property 'GPasteDeleteButton' on the module 'deleteButton'. That proper>
May 23 12:38:19 mars.localdomain gnome-shell[3124]: Some code accessed the property 'GPastePageItem' on the module 'pageItem'. That property was d>
May 23 12:38:19 mars.localdomain gnome-shell[3124]: clutter_paint_volume_set_width: assertion 'width >= 0.0f' failed
May 23 12:38:19 mars.localdomain gnome-shell[3124]: GNOME Shell started at Thu May 23 2019 12:38:18 GMT-0700 (PDT)
May 23 12:38:20 mars.localdomain gnome-shell[3124]: Activities Configurator Enabled
May 23 12:38:20 mars.localdomain gnome-shell[3124]: donotdistrub-button@nls1729 extension enabled.
May 23 12:38:36 mars.localdomain org.gnome.Shell.desktop[3124]: Window manager warning: Buggy client sent a _NET_ACTIVE_WINDOW message with a time>
```

I tried rerunning `sensors-detect --auto` just in case, but that didn't help.

I also tried installing Freon 37 from the extensions.gnome.org release (via GNOME Software) to rule out possible issues specific to my package, but it's the same problem.

UshakovVasilii, 2019-05-24:
Should be fixed by https://github.com/UshakovVasilii/gnome-shell-extension-freon/commit/a006cea782f25595e91dcf5853f97535983ac4db
I'll create new release  extensions.gnome.org

closed 2019-05-24 by UshakovVasilii

terrycloth, 2019-05-24:
Yup, that fixed it. Thanks!