UshakovVasilii/gnome-shell-extension-freon #136: AMD GPU with open source drivers

tpimh, 2019-05-28:
I'm using AMD GPU, but not using Catalyst. This is the information that `lm-sensors` report:
```
amdgpu-pci-0400
Adapter: PCI adapter
vddgfx:       +1.01 V  
fan1:        3997 RPM  (min =    0 RPM, max = 7000 RPM)
temp1:        +64.0°C  (crit = +97.0°C, hyst = -273.1°C)
power1:       16.16 W  (cap =  35.00 W)
```
However, Freon completely ignores it.

UshakovVasilii, 2019-05-30:
@tpimh please provide `sensors -j` output

tpimh, 2019-06-03:
Finally back to my machine. Full output is very long, only posting relevant part:
```json
   "amdgpu-pci-0400":{
      "Adapter": "PCI adapter",
      "vddgfx":{
         "in0_input": 0.875
      },
```

UshakovVasilii, 2019-06-05:
I checked provided output https://github.com/UshakovVasilii/gnome-shell-extension-freon/blob/i136/freon%40UshakovVasilii_Github.yahoo.com/sensorsUtil.js#L45-L65

Extension handle output correctly: 
[image]

tpimh, 2019-06-05:
Cool. It actually shows it, but I just didn't think that it was my graphics card. Thanks for pointing out.

closed 2019-06-05 by tpimh