UshakovVasilii/gnome-shell-extension-freon #137: Add nvme-cli as an option to get SSD temperature

tpimh, 2019-06-08:
I can get my SSD temperature with `nvme smart-log /dev/nvme0 | grep "^temperature"`. The utility comes from [this repo](https://github.com/linux-nvme/nvme-cli). I couldn't get Hddtemp, UDisks2 or smartctl to show the temperature of my SSD.

tpimh, 2019-06-08:
Here is the full output of smart-log:
```
Smart Log for NVME device:nvme0 namespace-id:ffffffff
critical_warning                    : 0
temperature                         : 43 C
available_spare                     : 100%
available_spare_threshold           : 10%
percentage_used                     : 0%
data_units_read                     : 1 198 745
data_units_written                  : 2 523 894
host_read_commands                  : 20 201 202
host_write_commands                 : 20 110 175
controller_busy_time                : 152
power_cycles                        : 371
power_on_hours                      : 294
unsafe_shutdowns                    : 164
media_errors                        : 0
num_err_log_entries                 : 0
Warning Temperature Time            : 0
Critical Composite Temperature Time : 0
Temperature Sensor 1                : 43 C
Temperature Sensor 2                : 49 C
Thermal Management T1 Trans Count   : 0
Thermal Management T2 Trans Count   : 0
Thermal Management T1 Total Time    : 0
Thermal Management T2 Total Time    : 0
```

UshakovVasilii, 2019-06-09:
related to #62

closed 2019-07-24 by UshakovVasilii

dsvensson, 2019-11-19:
Is there any way to get the temperature from a service rather than directly from the cli? At least on Ubuntu nvme-cli has no permission to access the drive as a normal user.

daenney, 2020-02-04:
@dsvensson As of Linux 5.5 this gets reported through the regular hwmon infrastructure that's part of sysfs. But until your distribution ships that kernel, you'll need to run a utility like nvme-cli which unfortunately has to be run as root.