UshakovVasilii/gnome-shell-extension-freon #138: Blacklist some of the sensors

tpimh, 2019-06-09:
There are some sensors shown that either show 0 or maximum value or other values that I just don't care about. They just take up space and also affect maximum and average temperature calculations. It would be nice to choose which sensors are shown (and maybe also rename them).

UshakovVasilii, 2019-06-10:
I think it should be done on lm_sensors side https://github.com/UshakovVasilii/gnome-shell-extension-freon/wiki/FAQ#system-to-hang-every-n-second

jonasmalacofilho, 2020-02-14:
The FAQ now has specific topics related to this:

 - [How do I rename sensors in the list?](https://github.com/UshakovVasilii/gnome-shell-extension-freon/wiki/FAQ#how-do-i-rename-sensors-in-the-list)
 - [How do I hide sensors from the list?](https://github.com/UshakovVasilii/gnome-shell-extension-freon/wiki/FAQ#how-do-i-hide-sensors-from-the-list)

---

@UshakovVasilii I think this can be closed too.

closed 2020-02-19 by UshakovVasilii