UshakovVasilii/gnome-shell-extension-freon #141: Freon seems to fail to load on Debian Sid

russel, 2019-08-13:
Debian Sid has GnomeShell 3.20.2, I have lm-sensors 3.5.0 and udisks2 2.8.4 installed. I am using Freon from a Git clone. I am using a Wayland login. It seems that Freon will not load (Tweak reports "Error loading extension") and on trying to get the preferences "There was an error loading the preferences dialog for Freon":
```
TypeError: ExtensionUtils.initTranslations is not a function

Stack trace:
  init@/home/users/russel/.local/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/prefs.js:17:5
  _getExtensionPrefsModule@resource:///org/gnome/shell/extensionPrefs/main.js:68:9
  wrapper@resource:///org/gnome/gjs/modules/_legacy.js:82:22
  _selectExtension@resource:///org/gnome/shell/extensionPrefs/main.js:82:31
  wrapper@resource:///org/gnome/gjs/modules/_legacy.js:82:22
  _onCommandLine@resource:///org/gnome/shell/extensionPrefs/main.js:235:17
  wrapper@resource:///org/gnome/gjs/modules/_legacy.js:82:22
  main@resource:///org/gnome/shell/extensionPrefs/main.js:389:5
  @<main>:1:43
```
I am not sure what I can do to find out any more detail or get this working.

UshakovVasilii, 2019-08-16:
You should use 34 version 
[image]

closed 2019-08-16 by UshakovVasilii

russel, 2019-08-16:
I switched to EOG-35 and everything works again. Thanks.