UshakovVasilii/gnome-shell-extension-freon #144: No icon on panel

gjm, 2019-09-18:
Hi.

Suddenly Freon stopped showing the temperature and icon on top panel. 

[image]

Sensors seem to be working since the correct values appear on the dropdown menu.

[image]

jonasmalacofilho, 2019-09-18:
That means that it can no longer read that sensor (see #122 and #124)

Maybe the driver for it was unloaded (or failed to load), maybe the sensor was simply renamed.  The first step to figuring this out would be to check the output of `sensors`.

gjm, 2019-09-18:
The output of `sensors` seems OK to me

```
dell_smm-virtual-0
Adapter: Virtual device
fan1:        3189 RPM
fan2:        3163 RPM

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +25.0°C  (crit = +107.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +55.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +53.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +54.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +54.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +53.0°C  (high = +100.0°C, crit = +100.0°C)

pch_skylake-virtual-0
Adapter: Virtual device
temp1:        +61.5°C  
```

What do I look for?

jonasmalacofilho, 2019-09-18:
The thing is, what sensor did you used to have up there?

gjm, 2019-09-18:
Nevermind... I figured it out... I just had to select which sensor(s) to display... Clear case of user error! :smile:

AlexWayfer, 2019-09-18:
Maybe for such cases we should display a hint like "Please select one or more sensors to display them on panel".

tpimh, 2019-10-30:
Just updated to Fedora 31 and I also have this bug, the panel looks like this:
[image]
I can't select sensors by clicking on them. Nothing happens on click. No errors are reported in `lg`, is there another error log somewhere?

jonasmalacofilho, 2019-10-30:
@tpimh that's probably _not_ the same problem.

What version of Gnome Shell are you running?

tpimh, 2019-10-30:
I'm using GNOME Shell 3.34.1.
If it's really a different problem, I can create another issue to track it.

jonasmalacofilho, 2019-10-30:
Weird, the extension seems to be working fine with 3.34.1.

Can you check what version of the extension you have?

Also, have you checked the output of `sensors`?

---

> If it's really a different problem, I can create another issue to track it.

I think that's a good idea.  You have the normal "no sensor selected" icon, not the "sensor fail" icon, and you can't select sensors, both of which seem very different than this.

tpimh, 2019-10-30:
I have created #154. Let's continue the discussion there.

closed 2019-11-09 by UshakovVasilii