UshakovVasilii/gnome-shell-extension-freon #146: Broken on 3.34

otaj, 2019-09-19:
Hi,

Today I updated gnome-shell to 3.34 and checked, whether something stopped working - and your extension was the only one.

Error from looking glass is:
```
Error: Wrong type undefined; object GIcon expected
```

Sadly, I don't have enough experience debugging GS extensions (and JS in general), so I don't know what other steps should I take.

UshakovVasilii, 2019-09-22:
Looks like getter:
```
    get gicon() {
        return this._gIcon;
    }
```
Is not called after `item.gicon`. I don't know what is wrong with gjs yet

jonasmalacofilho, 2019-09-25:
Possibly related to jose1711/gnome-shell-extension-sensory-perception@66542326660f769f99a5890bb927146a88664b74.

UshakovVasilii, 2019-09-25:
offtopic: TopIcons-plus not work, I cannot use Parcellite, I cannot switch language by Shift+Alt. I liked GNOME, but it unusable now  Xfce? )))

kouros17, 2019-09-26:
@UshakovVasili you can use KStatusNotifierItem/AppIndicator Support

UshakovVasilii, 2019-09-29:
Is extension work on 3.34 now?

jonasmalacofilho, 2019-09-29:
It's working on my laptop with 3.34.

What issues are you seeing?

UshakovVasilii, 2019-09-29:
Thanks! I will release it on extensions.gnome.org . I cannot check it, I'm using Xfce now.

closed 2019-09-29 by UshakovVasilii

jonasmalacofilho, 2019-09-29:
Off-topic: how are you liking Xfce these days/after coming from a recent Gnome?

UshakovVasilii, 2019-09-29:
@jonasmalacofilho  I miss windows live preview (( I very like it in gnome-shell. Xfce has Xfdashboard but it is not same.
Also I like this extension more than Xfce temperature applet))
Other things good. (I can switch keyboard layout by Shift-alt, I can open terminator from file manager, DE is lite, stable and configurable. I had problem with WICD, I installed NetworkManager instead, Xfce work good with NetworkManager)

jonasmalacofilho, 2019-09-29:
@UshakovVasilii thanks!

PR #152 (2019-09-27): Fix issues with GNOME 3.4
GNOME/gjs@72062b5e0368 ("object: Throw if constructing an unregistered object inheriting from GObject") started to enforce proper GObject inheritance through registerClass:

> When a class inherits from a GObject class, it must be always registered using GObject.registerClass, however gjs was accepting this silently [...]

Closes: #146 ("Broken on 3.34")  
See also: [GNOME/gjs/doc/Mapping.md]

[GNOME/gjs/doc/Mapping.md]: https://gitlab.gnome.org/GNOME/gjs/blob/57ba26807271a9b4fdd0f4bb5a8c9bad776d98d4/doc/Mapping.md
commit b821805f (2019-09-27): Construct FreonItem through GObject.registerClass

GNOME/gjs@72062b5e0368 ("object: Throw if constructing an unregistered
object inheriting from GObject") started to enforce proper GObject
inheritance through registerClass:

> When a class inherits from a GObject class, it must be always registered
> using GObject.registerClass, however gjs was accepting this silently [...]

Closes: #146 ("Broken on 3.34")
See also: [GNOME/gjs/doc/Mapping.md]

[GNOME/gjs/doc/Mapping.md]: https://gitlab.gnome.org/GNOME/gjs/blob/57ba26807271a9b4fdd0f4bb5a8c9bad776d98d4/doc/Mapping.md