UshakovVasilii/gnome-shell-extension-freon #154: Can not select sensors after system update

tpimh, 2019-10-30:
After updating to Fedora 31, the panel looks like this:
[image]
I can't select sensors by clicking on them. Nothing happens on click.

I'm using GNOME Shell 3.34.1 and sensors version 3.5.0 with libsensors version 3.5.0.

The extension is installed through extensions.gnome.org, I tried removing it and reinstalling the extension. The extension version is 39.

After resetting `org.gnome.shell.extensions.freon` recursivelly, it showed two random values until I clicked a sensor. After this it only showed the icon.

`sensors` output:
```
nct6779-isa-0a10
Adapter: ISA adapter
Vcore:                  +0.74 V  (min =  +0.00 V, max =  +1.74 V)
in1:                    +0.00 V  (min =  +0.00 V, max =  +0.00 V)
AVCC:                   +3.34 V  (min =  +2.98 V, max =  +3.63 V)
+3.3V:                  +3.34 V  (min =  +2.98 V, max =  +3.63 V)
in4:                    +0.00 V  (min =  +0.00 V, max =  +0.00 V)
in5:                    +0.00 V  (min =  +0.00 V, max =  +0.00 V)
in6:                    +1.07 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
3VSB:                   +3.30 V  (min =  +2.98 V, max =  +3.63 V)
Vbat:                   +3.15 V  (min =  +2.70 V, max =  +3.63 V)
in9:                    +1.03 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                   +0.16 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                   +0.14 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                   +1.03 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                   +1.58 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                   +0.18 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                     0 RPM  (min =    0 RPM)
fan2:                  2209 RPM  (min =    0 RPM)
fan3:                     0 RPM  (min =    0 RPM)
fan4:                     0 RPM  (min =    0 RPM)
fan5:                     0 RPM  (min =    0 RPM)
SYSTIN:                +114.0°C  (high =  +0.0°C, hyst =  +0.0°C)  ALARM  sensor = thermistor
CPUTIN:                +127.5°C  (high = +80.0°C, hyst = +75.0°C)  ALARM  sensor = CPU diode
AUXTIN0:                +22.5°C    sensor = thermistor
AUXTIN1:               +104.0°C    sensor = thermistor
AUXTIN2:               +106.0°C    sensor = thermistor
AUXTIN3:                +12.0°C    sensor = thermal diode
PECI Agent 0:           +33.0°C  
PCH_CHIP_CPU_MAX_TEMP:   +0.0°C  
PCH_CHIP_TEMP:           +0.0°C  
PCH_CPU_TEMP:            +0.0°C  
intrusion0:            OK
intrusion1:            ALARM
beep_enable:           disabled

amdgpu-pci-0400
Adapter: PCI adapter
vddgfx:       +0.88 V  
fan1:        3805 RPM  (min =    0 RPM, max = 7000 RPM)
edge:         +51.0°C  (crit = +97.0°C, hyst = -273.1°C)
power1:        7.07 W  (cap =  35.00 W)

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +47.0°C  (high = +71.0°C, crit = +81.0°C)
Core 0:        +39.0°C  (high = +71.0°C, crit = +81.0°C)
Core 1:        +36.0°C  (high = +71.0°C, crit = +81.0°C)
Core 2:        +46.0°C  (high = +71.0°C, crit = +81.0°C)
Core 3:        +38.0°C  (high = +71.0°C, crit = +81.0°C)
Core 4:        +39.0°C  (high = +71.0°C, crit = +81.0°C)
Core 8:        +42.0°C  (high = +71.0°C, crit = +81.0°C)
Core 9:        +43.0°C  (high = +71.0°C, crit = +81.0°C)
Core 10:       +44.0°C  (high = +71.0°C, crit = +81.0°C)
Core 11:       +44.0°C  (high = +71.0°C, crit = +81.0°C)
Core 12:       +44.0°C  (high = +71.0°C, crit = +81.0°C)
```

jonasmalacofilho, 2019-11-04:
I tested this on a fresh Fedora 31 install and still couldn't reproduce it.

Maybe try checking your journal for errors on other extensions or gnome-shell itself?

tpimh, 2019-11-05:
I tried removing `.config` and `.local` then reinstalling all extensions that I used and it works now. I don't know for sure what was causing it and can't reproduce it anymore, so closing the issue. Reopen if you can find out what can possibly cause it.

closed 2019-11-05 by tpimh

ManuelHu, 2019-11-20:
Maybe it helps someone who finds this issue while surfing on the web: For people where deleting `.config` and `.local` is no option, deleting the whole branch `/org/gnome/shell/extensions/freon` after fully uninstalling the extension did the trick for me. After a reboot and re-install it shows the temperature values in the top bar again.

(Maybe that was just a coincidence, but just reinstalling did not help...)

cristianlivella, 2020-03-22:
Same problem here, Fedora 31 and GNOME Shell 3.34.4. It worked well until a few minutes ago, now I touched something and I can no longer choose the sensor to show.