UshakovVasilii/gnome-shell-extension-freon #155: Extreme fan speeds

49studebaker, 2019-12-13:
See attached images.

[image]

[image]

UshakovVasilii, 2019-12-13:
Looks like it related to lm_sensors/drivers. You can execute `sensors` to check it.

closed 2019-12-13 by UshakovVasilii

jonasmalacofilho, 2019-12-13:
@49studebaker are you using an out-of-tree driver?

jonasmalacofilho, 2019-12-15:
@49studebaker 

Please respond to feedback/comments on your issues.  I was ready to ignore #156 and only give it a look because it reproduces on my system as well.

49studebaker, 2019-12-16:
https://bugs.launchpad.net/ubuntu/+source/lm-sensors/+bug/1856457