UshakovVasilii/gnome-shell-extension-freon #156: JS WARNING: [/usr/share/gnome-shell/extensions/ubuntu-dock@ubuntu.com/appIcons.js 1090]: unreachable code after return statement

49studebaker, 2019-12-15:
Ubuntu 19.10

Usage of object.actor is deprecated for Freon_FreonMenuButton
get@resource:///org/gnome/shell/ui/environment.js:249:29
_init@/home/htpc/.local/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/extension.js:96:9
enable@/home/htpc/.local/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/extension.js:637:17
_callExtensionEnable@resource:///org/gnome/shell/ui/extensionSystem.js:132:13
loadExtension@resource:///org/gnome/shell/ui/extensionSystem.js:264:21
_loadExtensions/<@resource:///org/gnome/shell/ui/extensionSystem.js:482:13
collectFromDatadirs@resource:///org/gnome/shell/misc/fileUtils.js:27:17
_loadExtensions@resource:///org/gnome/shell/ui/extensionSystem.js:457:9
_enableAllExtensions@resource:///org/gnome/shell/ui/extensionSystem.js:491:13
_sessionUpdated@resource:///org/gnome/shell/ui/extensionSystem.js:522:13
init@resource:///org/gnome/shell/ui/extensionSystem.js:32:9
_initializeUI@resource:///org/gnome/shell/ui/main.js:245:5
start@resource:///org/gnome/shell/ui/main.js:141:5
@<main>:1:31

jonasmalacofilho, 2019-12-15:
@UshakovVasilii 

Successfully reproduced on Gnome GNOME Shell 3.34.2; fix in linked PR.

49studebaker, 2019-12-16:
> @UshakovVasilii
> 
> Successfully reproduced on Gnome GNOME Shell 3.34.2; fix in linked PR.
> 
> @49studebaker
> 
> Is there any relation between ubuntu-dock and Freon or is the title of the issue simply wrong?

I do not know. The error was in my Ubuntu Log app.

PR #157 (2019-12-16): Remove deprecated use of Actor.actor
In Gnome 3.34 accessing `<object>.actor` results in a warning:

```
Usage of object.actor is deprecated for Freon_FreonMenuButton
get@resource:///org/gnome/shell/ui/environment.js:249:29
_init@[...]/freon@UshakovVasilii_Github.yahoo.com/extension.js:96:9
enable@[...]/freon@UshakovVasilii_Github.yahoo.com/extension.js:637:17
_callExtensionEnable@resource:///org/gnome/shell/ui/extensionSystem.js:148:13
loadExtension@resource:///org/gnome/shell/ui/extensionSystem.js:280:21
_loadExtensions/<@resource:///org/gnome/shell/ui/extensionSystem.js:490:13
collectFromDa [... cut]
commit dbcd29a5 (2019-12-16): Remove deprecated use of Actor.actor

In Gnome 3.34 accessing <object>.actor results in a warning:

Usage of object.actor is deprecated for Freon_FreonMenuButton
get@resource:///org/gnome/shell/ui/environment.js:249:29
_init@[...]/freon@UshakovVasilii_Github.yahoo.com/extension.js:96:9
enable@[...]/freon@UshakovVasilii_Github.yahoo.com/extension.js:637:17
_callExtensionEnable@resource:///org/gnome/shell/ui/extensionSystem.js:148:13
loadExtension@resource:///org/gnome/shell/ui/extensionSystem.js:280:21
_loadExtensions/<@resource:///org/gnome/shell/ui/extensionSystem.js:490:13
collectFromDatadir [... cut]
closed 2019-12-16 by UshakovVasilii