UshakovVasilii/gnome-shell-extension-freon #159: classify hwmon NVME temperature as drive temperature

daenney, 2020-02-04:
Starting with Linux 5.5 the hwmon infrastructure now reports `temp` for NVME drives. These currently show up using Freon but it would be nice to be able to categorise those as disk temperature items, like the ones from smartctl, hddtemp, udisks2 and nvmecli.

It seems two things would be needed:
* read `/sys/class/hwmon/hwmon*/name`, this will return `nvme` for NVME drives
* use the `drive-temperature` icon for these temperature sensors instead

daenney, 2020-02-04:
In my particular case sensors output has the following format:

```
nvme-pci-0100
Adapter: PCI adapter
Composite:
  temp1_{input,max,min,crit,alarm}
Sensor 1:
  temp2_{input,max,min}
Sensor 2:
  temp3_{input,max,min}
```

illwieckz, 2021-11-19:
It looks like #218 took care of that (for now the drive names are not displayed).

jonasmalacofilho, 2021-11-19:
That's true, I forgot to reference this issue.

The ~~drive~~ sensor names can be customized through `sensors.conf`.

daenney, 2021-11-19:
The icon would appear to be wrong. I'm getting it as a generic temperature sensor (which the thermometer icon) whereas my SATA drives show up with a drive icon. It's not terribly inconvenient but the classification does help when there's a bunch of sensors showing up.

jonasmalacofilho, 2021-11-19:
@daenney can you show me a screen shot?  And are any warnings or errors logged by freon/gnome-shell?

daenney, 2021-11-20:
This is what it looks like for me
[image]

```
$ sensors -j
   "nvme-pci-0100":{
      "Adapter": "PCI adapter",
      "NVMe: primary":{
         "temp1_input": 29.850,
         "temp1_max": 80.850,
         "temp1_min": -0.150,
         "temp1_crit": 81.850,
         "temp1_alarm": 0.000
      },
      "Sensor 1":{
         "temp2_input": 29.850,
         "temp2_max": 65261.850,
         "temp2_min": -273.150
      },
      "Sensor 2":{
         "temp3_input": 30.850,
         "temp3_max": 65261.850,
         "temp3_min": -273.150
      }
   }
```

Nothing stands out in logs.

jonasmalacofilho, 2021-11-21:
@daenney I can't see what could something like that (and can't reproduce it either).  What is the exact commit you're using (and have you made you don't have a dirty working tree)?

daenney, 2021-11-21:
Ah! I'm at #214 / commit: a11b1ea. It looks like #217 / #218 might be what I need to get this to go full circle.

That's my bad, let me rebuild and double check that.

daenney, 2021-11-21:
Right yup, that did it. Thanks for fixing this and my apologies for not catching I was on an older commit!

closed 2021-11-21 by daenney