UshakovVasilii/gnome-shell-extension-freon #160: [Feature request]: Hide sensors from the list

SebastianGrans, 2020-02-13:
As seen in the screenshot below, I have a lot of sensors which give out bogus reading (e.g. `AUXTIN0-3`). I'd like to be able to hide this from the list. 

It has been mentioned in other 'Issues' that you can remove these by adding `ignore` to the `sensors3.conf`, but I am unable to find these sensor labels in the conf file. 

From `sensors-detect` and `sensors` I can tell they originate from
```
Driver `nct6775':
  * ISA bus, address 0x290
    Chip `Nuvoton NCT6791D Super IO Sensors' (confidence: 9)
```
But even if I comment out that section from `sensors3.conf` they still show up. What ended up working was unloading the kernel module: `modprobe -r nct6775`.

[image]

[sensors3.txt](https://github.com/UshakovVasilii/gnome-shell-extension-freon/files/4197690/sensors3.txt)

jonasmalacofilho, 2020-02-13:
The `ignore` directive has to be used with "raw" sensor names. For example: `ignore temp1`.

To find out the correspondence between a sensor and it's label, try inspecting the output of `sensors -u`.

SebastianGrans, 2020-02-13:
This is part of what `sensors -u` gave me:
```
nct6791-isa-0290
Adapter: ISA adapter
[...]
AUXTIN0:
  temp3_input: 126.000
  temp3_type: 4.000
  temp3_offset: 0.000
AUXTIN1:
  temp4_input: -128.000
  temp4_type: 4.000
  temp4_offset: 0.000
[...]
```
I then tried adding:
```
ignore temp4_input
label temp3_input "Test"
``` 
under the heading `chip [...] "nct6791-*"` in my `sensors3.conf`. This didn't seem to change anything.
Do I have to reload the configuration somehow? `sensors -c /etc/sensors3.conf` didn't help.

jonasmalacofilho, 2020-02-13:
Try the following instead:

```
ignore temp4
label temp3 "Test"
```

Also, it's usually best to put your overrides and custom settings in a separate file.  Personally I like to use `/etc/sensors.d/custom.conf`.

SebastianGrans, 2020-02-14:
Ah, of course.🤦‍♂️
That works, thank you!

closed 2020-02-14 by UshakovVasilii

jonasmalacofilho, 2020-02-14:
> I'll add these instructions to the wiki in case other people have the same problems as me.
> **Edit: ** Done! [Link](https://github.com/UshakovVasilii/gnome-shell-extension-freon/wiki/FAQ#how-do-i-rename-sensors-in-the-list)

Good idea!