UshakovVasilii/gnome-shell-extension-freon #161: [Feature Request] Add support for NVIDIA PRIME render offload

s-kk, 2020-02-21:
Hello!
NVIDIA added support for PRIME render offload since version 435.17. In this mode, iGPU is selected by default. If users want to render a particular application with nvidia GPU, simply add some environment variables and run the executable.

Here's my setup:
OS: Arch Linux
NV Driver version: 440.59 with modesetting

I can directly grab thermal data with nvidia-smi command. That means, you can simply add a mode to use nvidia-smi and grab the thermal output.

Thanks in advance. BTW wonderful extension!

closed 2020-04-20 by s-kk

AlexWayfer, 2020-04-20:
> @s-kk closed this 35 minutes ago

Is it resolved?

s-kk, 2020-04-20:
> > @s-kk closed this 35 minutes ago
> 
> 
> 
> Is it resolved?

Yes. And it works as expected.

AlexWayfer, 2020-04-20:
> Yes. And it works as expected.

OK, thank you!