UshakovVasilii/gnome-shell-extension-freon #163: No NVMe temperature

OMGdaDPS, 2020-03-02:
Although I am able to change it to `nvme-cli` or `smartctl`, and both of them work in the terminal and produce output including the drives temperature, the extension does not show the drive when either is selected.

I seen over [here](https://github.com/UshakovVasilii/gnome-shell-extension-freon/issues/137#issuecomment-581931672) in another issue in your repo where someone said that using the `nvme-cli` one, it needs to be ran as root. How would you go about setting the extension to run it as root?

Any info that I can give you, please ask and I would be more than happy to provide it!

citizenu03bb, 2020-05-13:
I made it work by running `sudo visudo` to adjust (add it otherwise) the following line in `/etc/sudoers`:   
```conf
my_username     ALL = NOPASSWD: /usr/sbin/nvme list -o json, /usr/sbin/nvme smart-log /dev/nvme* -o json
```
and after editing `~/.local/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/nvmecliUtil.js`
```diff
--- ~/.local/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/nvmecliUtil.js.old
+++ ~/.local/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/nvmecliUtil.js
@@ -4,7 +4,7 @@
 
 function getNvmeData (argv){
     const nvme = GLib.find_program_in_path('nvme')
-    return JSON.parse(GLib.spawn_command_line_sync(`${nvme} ${argv} -o json`)[1].toString())
+    return JSON.parse(GLib.spawn_command_line_sync(`/usr/bin/sudo ${nvme} ${argv} -o json`)[1].toString())
 }
 
 var nvmecliUtil  = class {
```
I might have restarted the shell after that.

crysman, 2020-06-24:
confirming what @citizenu03bb proposes - in fresh 20.04 with gnome-shell 3.36.2 it works, too:

1. modifying sudoers via `sudo visudo` to allow that 2 nvme commands
1. modifying nvmecliUtil.js to actually invoke `/usr/bin/sudo...`

firasdib, 2020-07-02:
This could also be solved with a try catch, so the whole extension won't crash because of the JSON.parse.

waltyl14, 2020-07-05:
Hi all,

Would someone be willing to provide a bit more of a step-by-step for this process? I am relatively new to Linux, but would love to have Freon display NVMe temp correctly. Thanks for any help.

claudiopastorini, 2021-03-12:
For noobs like me you can allow the use of `nvme` command with sudo and without passowrd adding to the `sudoers` file (using `sudo visudo`) the following line:

```
yourusernamehere ALL = (root) NOPASSWD: /usr/bin/nvme
```

where `yourusernamehere` is your username

closed 2023-12-02 by UshakovVasilii