UshakovVasilii/gnome-shell-extension-freon #171: Change temperature SMART ID

aretaon, 2020-05-23:
I connected a newer Kingston SA400 SSD and Freon (with UDisks2 as temperature utility) reports 99 °C as temperature. For all other SSDs in the setup, temperature reporting works fine.

Using smartctl --all /dev/sda I get (some lines removed for clarity)

`ID# ATTRIBUTE_NAME          FLAG     VALUE WORST THRESH TYPE      UPDATED  WHEN_FAILED RAW_VALUE
194 Temperature_Celsius     0x0022   034   034   000    Old_age   Always       -       34 (Min/Max 23/34)
231 SSD_Life_Left           0x0000   099   099   000    Old_age   Offline      -       99
`
I assume that UDisks2 reports on SMART value 231 instead of 194 resulting in the unrealistic temperature read out.

Switching Freon to smartctl as sensor input did not help as no HDD/SSD temperature was shown using this setting. Is there a way to (1) change the SMART value that Freon reads from udisks2 or (2) switch the sensor readout to smartctl and define which SMART value is used?

corecontingency, 2020-06-03:
@aretaon This is an upsteam issue in `libatasmart` which udisks2 uses to read the temperate values.

Unfortunately, after making the library, Lennart Poettering moved on to bigger and better things, and upstream has been dead for over eight years. There are several outstanding patches for issues that no one has merged or looked at.

Here is the link for the bug report for this issue:
https://bugs.freedesktop.org/show_bug.cgi?id=97446

Luckily, this is one of the cases where the fix is simple and a patch has been made.
https://bugs.freedesktop.org/attachment.cgi?id=125967

So either compile and install libatasmart manually, or pester your distro maintainers to incorporate this patch.

aretaon, 2020-06-28:
Thanks @corecontingency for your help. So far I decided to keep with the official Ubuntu Kernel and and hddtemp.
Anyway concerning Freon, I consider this issue closed.

closed 2020-06-28 by aretaon