UshakovVasilii/gnome-shell-extension-freon #182: temperature is undefined.

1001101, 2020-08-08:
```
JS ERROR: Extension freon@UshakovVasilii_Github.yahoo.com: TypeError: getSmartData(...).temperature is undefined
get temp/<@/usr/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/smartctlUtil.js:38:34
get temp@/usr/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/smartctlUtil.js:35:35
_updateDisplay@/usr/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/extension.js:340:13
_updateUI@/usr/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/extension.js:296:18
_init@/usr/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/extension.js:119:14
enable@/usr/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/extension.js:641:17
```
It's bug.

renatose, 2021-02-23:
I have the same issue on Ubuntu 20.04

catvin82, 2024-03-05:
same on f39