UshakovVasilii/gnome-shell-extension-freon #184: nvme require sudo and breaks the application if not

blekinge, 2020-08-31:
Freon failed to run/install for me.

The log in journalctl showed
` journalctl _UID=1000 -e -l`
```
.yahoo.com: SyntaxError: JSON.parse: unexpected character at line 1 column 1 of the JSON data @ /home/abr/.local/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/nvmecliUtil.js:7
```

I found the relevant code in `nvmecliUtils.js`
```js
function getNvmeData (argv){
    const nvme = GLib.find_program_in_path('nvme')
    return JSON.parse(GLib.spawn_command_line_sync(`${nvme} ${argv} -o json`)[1].toString())
}
```

Running the same command from the command line produced

```
:) 12:19:30 abr@abr-pc:~$ nvme list -o json
12:20:44 'nvme list -o json'
Failed to open /dev/nvme0
{
  "Devices" : [
    {
      "NameSpace" : 0,
      "DevicePath" : "/dev/nvme0n1",
      "Firmware" : "        ",
      "Index" : 0,
      "ModelNumber" : "                                        ",
      "ProductName" : "Unknown device",
      "SerialNumber" : "                    ",
      "UsedBytes" : 0,
      "MaximumLBA" : 0,
      "PhysicalSize" : 0,
      "SectorSize" : 1
    }
  ]
}:) 12:20:44 abr@abr-pc:~$ 
```
So it will obviously try to parse "Failed to open /dev/nvme0" as JSON and die and take the entire Freon app down with it.

Now, I run with passwordless sudo (which is a mistake, but well...), so I tried to run
```
  12:22:51 abr@abr-pc:~$ sudo nvme list -o json
12:22:52 'sudo nvme list -o json'
{
  "Devices" : [
    {
      "NameSpace" : 1,
      "DevicePath" : "/dev/nvme0n1",
      "Firmware" : "EXA71HAQ",
      "Index" : 0,
      "ModelNumber" : "SAMSUNG MZVLB1T0HALR-000H2",
      "ProductName" : "Unknown device",
      "SerialNumber" : "S497NA0MC04922",
      "UsedBytes" : 742922313728,
      "MaximumLBA" : 2000409264,
      "PhysicalSize" : 1024209543168,
      "SectorSize" : 512
    }
  ]
}:) 12:22:52 abr@abr-pc:~$ 
```

and changed to code to
```js

function getNvmeData (argv){
    const nvme = GLib.find_program_in_path('nvme')
    return JSON.parse(GLib.spawn_command_line_sync(`sudo ${nvme} ${argv} -o json`)[1].toString())
}
```
to get nvme working and Freon working. And now it works. 

But perhaps include a warning or something for anyone else trying to use nvme without being root

ptomato, 2020-09-30:
I found you can work around this by choosing a different value for "HDD/SSD temperature utility" that isn't "nvme-cli", then removing and reinstalling Freon.

dansleboby, 2020-10-02:
[FREON] Unable to find nvme devices: SyntaxError: JSON.parse: unexpected end of data at line 1 column 1 of the JSON data

dansleboby@pop-os:~$ nvme list -o json
Failed to open /dev/nvme0n1: Permission denied

root@pop-os:~# nvme list -o json
{
  "Devices" : [
    {
      "NameSpace" : 1,
      "DevicePath" : "/dev/nvme0n1",
      "Firmware" : "15301229",
      "Index" : 0,
      "ModelNumber" : "PM981a NVMe SAMSUNG 2048GB",
      "ProductName" : "Unknown device",
      "SerialNumber" : "      S4UFNF0N508905",
      "UsedBytes" : 90264449024,
      "MaximumLBA" : 4000797360,
      "PhysicalSize" : 2048408248320,
      "SectorSize" : 512
    }
  ]
}

LAKostis, 2020-10-09:
But you can always create a shell wrapper for nvme-cli which actually calls it via sudo:
```
$ cat ./nvme 
#!/bin/bash -efu
sudo /usr/sbin/nvme $*
```
or use shell alias instead.

barneydesmond, 2021-03-15:
I have the same issue and don't know of an easy _and_ clean way around it. Freon needs access to that information which is only available through the `nvme` utility. I used the shell wrapper that @LAKostis mentioned above and it works perfectly.

For anyone looking to replicate that:
* Install the nvme-cli tool for your distro
* Create the wrapper script and put it at `/usr/local/bin/nvme`, make sure to set it executable
`chmod +x /usr/local/bin/nvme`
* Edit sudoers using `visudo` and give yourself access to run it without needing a password
`yourUsername ALL=(root) NOPASSWD: /usr/sbin/nvme`
* Open a new shell and test the command, you should be able to see the serial and model of your NVMe drives without needing to enter a password
`nvme list`

Now Freon should be able to read your NVMe drive status info, once you select it in the Freon settings.

bBSempai, 2021-07-25:
So I tried all of these methods by editing the `sudoers` and  `nvmecliUtils.js`. But still to no avail. I can definitely get the SSD temperature through TLP and stuff but not able to display it on the freon along with my HDD. Can see only HDD. Can anyone help ;(

[image]

illwieckz, 2023-12-15:
This could be ported to use the pkexec mechanisms… once this pkexec mechanisms is fixed (#274), it means someone has to write a policy file to ship and write the plumbing to make the extension asks the user to install the policy file if needed.