UshakovVasilii/gnome-shell-extension-freon #186: Error `meta is null` on Gnome 3.38

MonsieurV, 2020-09-29:
The extension does not work anymore on Gnome 3.38.

Stack trace:

```
(gnome-shell:9657): Gjs-WARNING **: 13:14:26.483: JS ERROR: Extension freon@UshakovVasilii_Github.yahoo.com: TypeError: meta is null
_patchContainerClass/containerClass.prototype.child_set@resource:///org/gnome/shell/ui/environment.js:43:13
_patchContainerClass/containerClass.prototype.add@resource:///org/gnome/shell/ui/environment.js:52:18
_appendStaticMenuItems@/home/yoan/.local/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/extension.js:481:20
_updateDisplay@/home/yoan/.local/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/extension.js:472:18
_updateUI@/home/yoan/.local/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/extension.js:292:18
_init@/home/yoan/.local/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/extension.js:119:14
enable@/home/yoan/.local/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/extension.js:637:17
_callExtensionEnable@resource:///org/gnome/shell/ui/extensionSystem.js:167:32
loadExtension@resource:///org/gnome/shell/ui/extensionSystem.js:348:26
callback@resource:///org/gnome/shell/ui/extensionDownloader.js:222:39
gotExtensionZipFile/<@resource:///org/gnome/shell/ui/extensionDownloader.js:106:13
```

jonasmalacofilho, 2020-09-30:
I'm not on 3.38 yet, but this problem is probably fixed by #183 (already merged into the master branch).

MonsieurV, 2020-10-01:
Indeed this work by manual installation from the master branch.

Could there be a release on https://extensions.gnome.org/extension/841/freon/?

UshakovVasilii, 2020-10-01:
Done, under review

mokilcde, 2020-10-03:
Hi!

I have to add also this patch to get it to work:

```
--- /tmp/freonItem.js	2020-10-03 05:46:53.555147052 +0200
+++ freonItem.js	2020-10-03 05:47:41.939514289 +0200
@@ -1,6 +1,7 @@
 const GObject = imports.gi.GObject;
 const PopupMenu = imports.ui.popupMenu;
 const St = imports.gi.St;
+const Clutter = imports.gi.Clutter;
 
 var FreonItem = GObject.registerClass(class FreonItem extends PopupMenu.PopupBaseMenuItem {
 
@@ -10,9 +11,9 @@ var FreonItem = GObject.registerClass(cl
         this._key = key;
         this._gIcon = gIcon;
 
-        this._labelActor = new St.Label({text: displayName ? displayName : label});
+        this._labelActor = new St.Label({text: displayName ? displayName : label, x_align: Clutter.ActorAlign.CENTER, x_expand: true});
         this.actor.add(new St.Icon({ style_class: 'popup-menu-icon', gicon : gIcon}));
-        this.actor.add(this._labelActor, {x_fill: true, expand: true});
+        this.actor.add_child(this._labelActor);
         this._valueLabel = new St.Label({text: value});
         this.actor.add(this._valueLabel);
     }
```

UshakovVasilii, 2020-10-03:
Looks like latest update work on gnome-shell 3.38 only. I inactivated latest update and uploaded new version for 3.38 only.

[image]

janosbrodbeck, 2020-10-03:
I can confirm what @mokilcde posted. Without his patch the extensioned worked somehow, but I couldn't open the popup-menu when I clicked on the item in the panel.
 
Its working fine now with his patch.

uno20001, 2020-10-05:
I got the following error:

```
JS ERROR: TypeError: meta is null
_patchContainerClass/containerClass.prototype.child_set@resource:///org/gnome/shell/ui/environment.js:43:13
_patchContainerClass/containerClass.prototype.add@resource:///org/gnome/shell/ui/environment.js:52:18
_init@/usr/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/freonItem.js:15:20
_appendMenuItems@/usr/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/extension.js:528:28
_updateDisplay@/usr/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/extension.js:462:22
_updateUI@/usr/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/extension.js:297:18
_init/this._updateUITimeoutId<@/usr/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/extension.js:122:18

```
and I can confirm that @mokilcde's patch works.

jonasmalacofilho, 2020-10-06:
Just upgraded to 3.38 myself.  I found that @mokilcde's patch messed up the aligned on the Freon items, so I slightly changed it to:

```diff
diff --git a/freon@UshakovVasilii_Github.yahoo.com/freonItem.js b/freon@UshakovVasilii_Github.yahoo.com/freonItem.js
index 05d623da88ed..901e0e4a4608 100644
--- a/freon@UshakovVasilii_Github.yahoo.com/freonItem.js
+++ b/freon@UshakovVasilii_Github.yahoo.com/freonItem.js
@@ -10,9 +10,9 @@ var FreonItem = GObject.registerClass(class FreonItem extends PopupMenu.PopupBas
         this._key = key;
         this._gIcon = gIcon;
 
-        this._labelActor = new St.Label({text: displayName ? displayName : label});
+        this._labelActor = new St.Label({text: displayName ? displayName : label, x_expand: true});
         this.actor.add(new St.Icon({ style_class: 'popup-menu-icon', gicon : gIcon}));
-        this.actor.add(this._labelActor, {x_fill: true, expand: true});
+        this.actor.add_child(this._labelActor);
         this._valueLabel = new St.Label({text: value});
         this.actor.add(this._valueLabel);
     }
```

<details><summary>Expand for diff on top of @mokilcde's patch.</summary>

```diff
diff --git a/freon@UshakovVasilii_Github.yahoo.com/freonItem.js b/freon@UshakovVasilii_Github.yahoo.com/freonItem.js
index 84c4c8989ccf..901e0e4a4608 100644
--- a/freon@UshakovVasilii_Github.yahoo.com/freonItem.js
+++ b/freon@UshakovVasilii_Github.yahoo.com/freonItem.js
@@ -1,7 +1,6 @@
 const GObject = imports.gi.GObject;
 const PopupMenu = imports.ui.popupMenu;
 const St = imports.gi.St;
-const Clutter = imports.gi.Clutter;
 
 var FreonItem = GObject.registerClass(class FreonItem extends PopupMenu.PopupBaseMenuItem {
 
@@ -11,7 +10,7 @@ var FreonItem = GObject.registerClass(class FreonItem extends PopupMenu.PopupBas
         this._key = key;
         this._gIcon = gIcon;
 
-        this._labelActor = new St.Label({text: displayName ? displayName : label, x_align: Clutter.ActorAlign.CENTER, x_expand: true});
+        this._labelActor = new St.Label({text: displayName ? displayName : label, x_expand: true});
         this.actor.add(new St.Icon({ style_class: 'popup-menu-icon', gicon : gIcon}));
         this.actor.add_child(this._labelActor);
         this._valueLabel = new St.Label({text: value});
```
</details>

UshakovVasilii, 2020-10-08:
Please create pull request, I cannot check it, I'm using Xfce now

mokilcde, 2020-10-08:
Done #188

gilvbp, 2020-10-08:
> Just upgraded to 3.38 myself. I found that @mokilcde's patch changed the alignment on the Freon items, so I slightly changed it to:
> 
> ```diff
> diff --git a/freon@UshakovVasilii_Github.yahoo.com/freonItem.js b/freon@UshakovVasilii_Github.yahoo.com/freonItem.js
> index 05d623da88ed..901e0e4a4608 100644
> --- a/freon@UshakovVasilii_Github.yahoo.com/freonItem.js
> +++ b/freon@UshakovVasilii_Github.yahoo.com/freonItem.js
> @@ -10,9 +10,9 @@ var FreonItem = GObject.registerClass(class FreonItem extends PopupMenu.PopupBas
>          this._key = key;
>          this._gIcon = gIcon;
>  
> -        this._labelActor = new St.Label({text: displayName ? displayName : label});
> +        this._labelActor = new St.Label({text: displayName ? displayName : label, x_expand: true});
>          this.actor.add(new St.Icon({ style_class: 'popup-menu-icon', gicon : gIcon}));
> -        this.actor.add(this._labelActor, {x_fill: true, expand: true});
> +        this.actor.add_child(this._labelActor);
>          this._valueLabel = new St.Label({text: value});
>          this.actor.add(this._valueLabel);
>      }
> ```

> 
> Expand for diff on top of @mokilcde's patch.
Nice, now it's working, but still missing item text labels. Only showing temps on the list.

jonasmalacofilho, 2020-10-08:
@gilvbp that's curious, text labels are showing fine for me (and I imagine for the others that posted here).

Can you post a screenshot and take a look at the logs?

gilvbp, 2020-10-08:
> @gilvbp that's curious, text labels are showing fine for me (and I imagine for the others that posted here).
> 
> Can you post a screenshot and take a look at the logs?

I double-check the .js and saw that was missing `const Clutter = imports.gi.Clutter;`. Now it's ok. Obrigado, doutor! Tmj

jonasmalacofilho, 2020-10-08:
@gilvbp great!

Although that's only necessary if you have kept `x_align: Clutter.ActorAlign.CENTER`.

spxak1, 2020-11-01:
Is a newer version that works in 3.38 available now? How can we get it? Thanks.

UshakovVasilii, 2020-11-01:
@spxak1 you can use https://extensions.gnome.org/extension/841/freon/

spxak1, 2020-11-01:
> @spxak1 you can use https://extensions.gnome.org/extension/841/freon/

Thanks, but this is what I get:

`JS ERROR: Extension freon@UshakovVasilii_Github.yahoo.com: SyntaxError: JSON.parse: unexpected character at line 1 column 1 of the JSON data @ /home/otheos/.local/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/nvmecliUtil.js:7`

mokilcde, 2020-11-02:
Hi!

The merge request is already pushed so it shall be available.

But you can chekcout the git version and see whether it works

CU
  Michael

> Is a newer version that works in 3.38 available now? How can we get it? Thanks.
CU  
         
  Michael

closed 2020-12-08 by MonsieurV