UshakovVasilii/gnome-shell-extension-freon #196: Support in-kernel drivetemp

linusw, 2021-03-21:
The kernel hardware monitor subsystem nowadays has a module to monitor harddrives as any other hwmon sensor.
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/hwmon/drivetemp.c

This will be available after "modprobe drivetemp" which is something you can add by default to e.g. Fedora by adding a file named /etc/modules-load.d/drivetemp with a single line saying "drivetemp".

"sensors" was augmented to deal with this and looks like so:

# sensors
drivetemp-scsi-1-10
Adapter: SCSI adapter
temp1:        +31.0°C  (low  =  +0.0°C, high = +60.0°C)
                       (crit low = -41.0°C, crit = +85.0°C)
                       (lowest = +24.0°C, highest = +32.0°C)

It would be nice if Freon could support this out of the box as well so no extra hddtemp etc tools are needed.

illwieckz, 2021-11-19:
See #217 and #218, `drivetemp` and `nvme` sensors are now recognized as drives. The names are not displayed though.

linusw, 2022-03-02:
> See #217 and #218, `drivetemp` and `nvme` sensors are now recognized as drives. The names are not displayed though.

Excellent work! Closing this because I am happy.

closed 2022-03-02 by linusw