UshakovVasilii/gnome-shell-extension-freon #199: It causes high vsync-jitter for MPV player on Gnome40 Wayland.

laichiaheng, 2021-04-26:
```                   -`                    
                  .o+`                   --------------------- 
                 `ooo/                   OS: Arch Linux x86_64 
                `+oooo:                  Kernel: 5.11.16-arch1-1 
               `+oooooo:                 Uptime: 2 hours, 41 mins 
               -+oooooo+:                Packages: 1103 (pacman) 
             `/:-:++oooo+:               Shell: zsh 5.8 
            `/++++/+++++++:              Resolution: 1920x1080 
           `/++++++++++++++:             DE: GNOME 40.1 
          `/+++ooooooooooooo/`           WM: Mutter 
         ./ooosssso++osssssso+`          WM Theme: Adwaita 
        .oossssso-````/ossssss+`         Theme: Adwaita-dark [GTK2/3] 
       -osssssso.      :ssssssso.        Icons: Adwaita [GTK2/3] 
      :osssssss/        osssso+++.       Terminal: gnome-terminal 
     /ossssssss/        +ssssooo/-       CPU: AMD Ryzen 7 3700X (16) @ 3.600GHz 
   `/ossssso+/:-        -:/+osssso+-     GPU: AMD ATI Radeon RX 470/480/570/570X/580/580X/59 
  `+sso+:-`                 `.-/+oso:    Memory: 3207MiB / 32072MiB 
 `++:.                           `-/+/
 .`                                 `/                           
```
It causes high vsync-jitter while playing videos no Gnome40 Wayland, which has never happened on Gnome40 Xorg. If I disable this extension, the vsync-jitter can remain 0.

mvrk69, 2024-01-19:
I had similar issue on Fedora 37 (Xorg) and culprit was not exactly the extension, but the use of smartctl for disk sensors, as soon as i disabled it no more jitter/stutter on movie playback.

mvrk69, 2024-09-13:
Actually, now i'm on Fedora 40 (Xorg), and i have this problem even with smartctl disabled, when the extension is active causes stutter on video playback (youtube / mpv / vlc) i have to disable it when i want to play any video.

italoghost, 2024-09-24:
Experiencing this as well! It is possible to test it (if you have an VRR monitor) with VRR test.