UshakovVasilii/gnome-shell-extension-freon #208: Bug: Position in panel

agnath18K, 2021-07-09:
The icon loses it's position index in the panel after restarting pc. It's not keeping the defined position. Please find the screenshot attached.
[image]
[image]

bBSempai, 2021-07-25:
Same problem with me. Loses its position after restart.

After restart screenshot:  (originally should have been at the left most end)
[image]

xproot, 2021-08-06:
this also happens to me

BiNh0X, 2022-05-02:
Same here.

Ubuntu 20.04.4

Gnome 3.36.8