UshakovVasilii/gnome-shell-extension-freon #223: On the pertinence of using combobox lists when it may be useful to merge the lists

illwieckz, 2021-11-19:
Some sensor provider selection are made the combobox way, you can for example choose between `hddtemp` and `nvme-cli`, but not use both. But  I don't have NVME drives right now but I would be surprised if `nvme-cli` can list temperatures for non-NVME drive for example. Another example is the GPU provider selection, one may have more than one GPU, and GPUs of different brands. Nvidia may defeat the ability to run various GPUs from various brands as they usually replace half the system, but that's not Freon's fault.

Maybe it would be better if all those various sensor providers would just be listed with a checkbox to enable or disable each of them, and get the provided data from of any them concatenated in the final list to display.

jonasmalacofilho, 2021-11-19:
Seconded, but I think they should be visibly grouped in some way.

Maybe three categories?

- general providers (lm-sensors, liquidctl, IPMI);
- GPU providers (NVIDIA/AMD utilities); and 
- drive providers (smartctl, etc.).

> I would be surprised if nvme-cli can list temperatures for non-NVME drive for example

Indeed it can't.

---

On a slight tangent, maybe drive providers should be deprecated (and, in the future, removed) in favor of the `drivetemp` and `nvme` support in HWMON?

(I already started to work on fixing the lm-sensors issue around selecting `drivetemp` "chips": so far I have proposed a patch making lm-sensors less restrictive; even if it isn't accepted as is, it's a starting point for discussing with the lm-sensors maintainers).

illwieckz, 2021-11-19:
> Seconded, but I think they should be visibly grouped in some way.
> Maybe three categories?
> - general providers (lm-sensors, liquidctl, IPMI);
> - GPU providers (NVIDIA/AMD utilities); and
> - drive providers (smartctl, etc.).

I thought about the same. Also lm-sensors may be turned optional as well.

> > I would be surprised if nvme-cli can list temperatures for non-NVME drive for example
>
> Indeed it can't.

So it's confirmed having to chose between those sensor providers is not useful in such case, then…

> On a slight tangent, maybe drive providers should be deprecated (and, in the future, removed) in favor of the `drivetemp` and `nvme` support in HWMON?

As long as they work, they can be kept I guess. There is the same question for Catalyst GPU sensor provider, it's for the now-deprecated fglrx driver that got replaced by amdgpu-pro long time ago (based on free amdgpu, then hwmon based). I still have a working setup with fglrx in order to check for compatibility and regression of some software I develop, and I have read last year some people still using fglrx for some OpenCL tasks on some GPU newer drivers doesn't have complete OpenCL support yet (or uneasy to get working), but that's really deprecated. In the same way I would say that as long as they work it can be kept, but that's clearly deprecated.

> (I already started to work on fixing the lm-sensors issue around selecting `drivetemp` "chips": so far I have proposed a patch making lm-sensors less restrictive; even if it isn't accepted as is, it's a starting point for discussing with the lm-sensors maintainers).

I have seen that, that's nice! I hope it will brings something good!

illwieckz, 2021-11-23:
#226 turned lists into independent switches.

closed 2021-11-23 by illwieckz