UshakovVasilii/gnome-shell-extension-freon #230: Gnome 42?

techyporcupine, 2022-03-25:
Can we get support for Gnome 42?
I am running fedora 36 with Gnome 42, and the extension is "outdated"

JmbFountain, 2022-03-25:
I just checked, and the Extension seems to install fine by cloning the git repo and installing manually.

retrixe, 2022-03-26:
Nonetheless, it would be a good idea to publish an update to https://extensions.gnome.org

UshakovVasilii, 2022-03-27:
Done, under review

UshakovVasilii, 2022-03-27:
Review failed:
> you cannot use `sudo` in extensions. You should use `pkexec` instead:
> https://gjs.guide/extensions/review-guidelines/review-guidelines.html#privileged-subprocess-must-not-be-user-writable

Related to #220

techyporcupine, 2022-04-07:
You can manually install.

Ud4ba, 2022-04-07:
> You can manually install.

Thanks.Will this get an update at https://extensions.gnome.org/ ?

retrixe, 2022-04-08:
I think it would be a good idea to temporarily revert the ipmi-sensors support and publish the 42 release without it, because it looks like you need elevated privileges to read from ipmi-sensors, and for that you'd ideally need some sort of background daemon running as an elevated user from which you can query the sensor data as a regular user. This would be a bit involved to setup, and if not this, then you would need some way to configure ipmi-sensors data to be readable as non-root, so it might be advisable to revert the support for now and publish an update until a solution can be found.

techyporcupine, 2022-04-08:
I agree with you @retrixe. I don't know how many people use `ipmi-sensors`, but if they truly need to, they can just clone from Github and install themselves.

techyporcupine, 2022-04-08:
@UshakovVasilii Is it possible to remove `ipmi-sensors` support then publish the update?

techyporcupine, 2022-04-08:
I would be willing to test the extension.

Its-Alex, 2022-04-11:
I would be willing to test the extension too :slightly_smiling_face:.

UshakovVasilii, 2022-04-12:
@techyporcupine @Its-Alex I created https://github.com/UshakovVasilii/gnome-shell-extension-freon/tree/without-ipmi branch, please test it

techyporcupine, 2022-04-12:
Just installed. So far it seems to be working, I can keep using it for a while and see if it breaks.

Ud4ba, 2022-04-13:
Thanks for the update works fine :)

techyporcupine, 2022-04-13:
Been using it for about a day, seems to be stable.

Its-Alex, 2022-04-13:
Tested on Archlinux that works 👍 Thanks for the update 🙏

UshakovVasilii, 2022-04-14:
Approved on extensions.gnome.org

closed 2022-04-14 by UshakovVasilii