UshakovVasilii/gnome-shell-extension-freon #232: AticonfigUtil is not a constructor

telometto, 2022-04-25:
I'm getting the error mentioned in the title. I cannot run the extension; I've tried uninstalling it, reinstalling it, removing the folder etc. but to no avail.

telometto, 2022-04-25:
Easy fix. Disable Catalyst if you do not use the proprietary drivers and log out or reboot.

closed 2022-04-25 by telometto

bulgakke, 2022-06-20:
> Need just a little more context here. What is catalyst. What command should we use to disable it? looks like freon wants to execute a utility called aticonfig, and that's not installed on my system. Shouldn't it check if binary exists, and display a warning if it does not? (As opposed to crashing and all info going away?)

Tick this flag in the extensions's settings.
[image]