UshakovVasilii/gnome-shell-extension-freon #239: AticonfigUtil is not a constructor

BillyCroan, 2022-05-24:
I hope you're doing well Vasilii.

I don't know exactly how I broke it, but I was trying to enable all possible sensors, and shortly thereafter, freon went away.  I looked in the gnome extensions control panel and I saw the error in red "AticonfigUtil+is+not+a+constructor".  Freon was off, and the control panel wouldn't let me turn it back on.

It looks like freon tries to execute a binary by the name of "aticonfig".  But I don't have that binary as I don't have any ATI equipment.

To get freon working again I had to comment out all lines referencing aticonfig.  Probably a better solution would be to check if the binary exists before trying to use it and simply hide ati related things if it's not there, or show a message to the user in the place of those settings, asking the user to go install it.

cd /home/yourusernamehere/.local/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/
grep -i -r aticonfig .
vim extensions.js
search for ticonf
comment all lines including aticonfig.
alt-F2 "r" to restart the shell

nightcrawler086, 2022-05-31:
this worked for me.

BillyCroan, 2022-06-03:
I thought it worked for me too and later that day started getting more errors from freon in /var/log/messages. did you?
I ended up settling for two plugins: system-monitor-next ( https://extensions.gnome.org/extension/3010/system-monitor-next/ ) gives you graphs of temp and fan over time.  And a plugin called "Executor" ( https://extensions.gnome.org/extension/2932/executor/ ) runs any arbitrary command and displays the stdout.  Then I used a short script to grab the highest temp and its name from 'sensors' and print it:
IFS=":"
sensors | grep '+.*C' | sed 's/:\ *+/:/;s/\..*//' | while read ID Temp
do
      echo "${Temp}°$( sed 's/Package/Pak/;s/id//;s/Other/oth/;s/\ //g;' <<<$ID )" 
done | sort -n | tail -n 1

nightcrawler086, 2022-06-03:
The extension is working, but I do some errors related to the extension:

`
Jun  2 08:24:57 rog gnome-shell[3539]: Some code called array.toString() on a Uint8Array instance. Previously this would have interpreted the bytes of the array as a string, but that is nonstandard. In the future this will return the bytes as comma-separated digits. For the time being, the old behavior has been preserved, but please fix your code anyway to explicitly call ByteArray.toString(array).#012(Note that array.toString() may have been called implicitly.)#0120 HddtempUtil() ["/home/nightcrawler/.local/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/hddtempUtil.js":27:114]#0121 _initHddtempUtility() ["/home/nightcrawler/.local/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/extension.js":372:34]#0122 _init() ["/home/nightcrawler/.local/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/extension.js":83:13]#0123 ButtonBox() ["resource:///org/gnome/shell/ui/panelMenu.js":11:0]#0124 PanelMenuButton() ["resource:///org/gnome/shell/ui/panelMenu.js":97:3]#0125 Freon_FreonMenuButton() ["/home/nightcrawler/.local/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/extension.js":57:46]#0126 enable() ["/home/nightcrawler/.local/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/extension.js":960:16]#0127 _callExtensionEnable() ["resource:///org/gnome/shell/ui/extensionSystem.js":182:31]#0128 loadExtension() ["resource:///org/gnome/shell/ui/extensionSystem.js":382:25]#0129 _loadExtensions/<() ["resource:///org/gnome/shell/ui/extensionSystem.js":638:17]#01210 collectFromDatadirs() ["resource:///org/gnome/shell/misc/fileUtils.js":27:27]#01211 _loadExtensions() ["resource:///org/gnome/shell/ui/extensionSystem.js":613:18]#01212 _enableAllExtensions() ["resource:///org/gnome/shell/ui/extensionSystem.js":644:17]#01213 _sessionUpdated() ["resource:///org/gnome/shell/ui/extensionSystem.js":664:13]#01214 init() ["resource:///org/gnome/shell/ui/extensionSystem.js":56:13]#01215 _initializeUI() ["resource:///org/gnome/shell/ui/main.js":298:21]#01216 start() ["resource:///org/gnome/shell/ui/main.js":175:4]#01217 <TOP LEVEL> ["resource:///org/gnome/shell/ui/init.js":6:16]#01218 InnerModuleEvaluation() ["self-hosted":2411:30]#01219 evaluation() ["self-hosted":2358:23]
Jun  2 08:24:57 rog gnome-shell[3539]: Some code called array.toString() on a Uint8Array instance. Previously this would have interpreted the bytes of the array as a string, but that is nonstandard. In the future this will return the bytes as comma-separated digits. For the time being, the old behavior has been preserved, but please fix your code anyway to explicitly call ByteArray.toString(array).#012(Note that array.toString() may have been called implicitly.)#0120 HddtempUtil() ["/home/nightcrawler/.local/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/hddtempUtil.js":38:74]#0121 _initHddtempUtility() ["/home/nightcrawler/.local/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/extension.js":372:34]#0122 _init() ["/home/nightcrawler/.local/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/extension.js":83:13]#0123 ButtonBox() ["resource:///org/gnome/shell/ui/panelMenu.js":11:0]#0124 PanelMenuButton() ["resource:///org/gnome/shell/ui/panelMenu.js":97:3]#0125 Freon_FreonMenuButton() ["/home/nightcrawler/.local/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/extension.js":57:46]#0126 enable() ["/home/nightcrawler/.local/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/extension.js":960:16]#0127 _callExtensionEnable() ["resource:///org/gnome/shell/ui/extensionSystem.js":182:31]#0128 loadExtension() ["resource:///org/gnome/shell/ui/extensionSystem.js":382:25]#0129 _loadExtensions/<() ["resource:///org/gnome/shell/ui/extensionSystem.js":638:17]#01210 collectFromDatadirs() ["resource:///org/gnome/shell/misc/fileUtils.js":27:27]#01211 _loadExtensions() ["resource:///org/gnome/shell/ui/extensionSystem.js":613:18]#01212 _enableAllExtensions() ["resource:///org/gnome/shell/ui/extensionSystem.js":644:17]#01213 _sessionUpdated() ["resource:///org/gnome/shell/ui/extensionSystem.js":664:13]#01214 init() ["resource:///org/gnome/shell/ui/extensionSystem.js":56:13]#01215 _initializeUI() ["resource:///org/gnome/shell/ui/main.js":298:21]#01216 start() ["resource:///org/gnome/shell/ui/main.js":175:4]#01217 <TOP LEVEL> ["resource:///org/gnome/shell/ui/init.js":6:16]#01218 InnerModuleEvaluation() ["self-hosted":2411:30]#01219 evaluation() ["self-hosted":2358:23]
Jun  2 08:24:57 rog gnome-shell[3539]: [FREON] Unable to find smart devices: GLib.SpawnError: Failed to execute child process “null” (No such file or directory)
Jun  2 08:24:57 rog gnome-shell[3539]: [FREON] Unable to find nvme devices: GLib.SpawnError: Failed to execute child process “null” (No such file or directory)
Jun  3 10:15:23 rog gnome-shell[3539]: [FREON] Unable to find smart devices: GLib.SpawnError: Failed to execute child process “null” (No such file or directory)
Jun  3 10:15:23 rog gnome-shell[3539]: [FREON] Unable to find nvme devices: GLib.SpawnError: Failed to execute child process “null” (No such file or directory)
`

It's still reporting temperatures for CPU/GPU/Disk so I'm hesitant to change it.

f8ttyc8t, 2022-07-27:
Astonishing tool, well done!

Unfortunately, same problem (AticonfigUtil is not a constructor) here - any help appreciated!
I'm having ATI card installed, but no aticonfig tool.

(Running Pop!_OS 22.04 LTS on MinisForum UM 700 being equipped with AMD Radon Vega 10 Graphics)

PiemP, 2022-08-25:
Is enough to comment this line of code in the 'extension.js' file:

https://github.com/UshakovVasilii/gnome-shell-extension-freon/blob/dab42ffc5c8405a74f835316acb38c32474d2243/freon%40UshakovVasilii_Github.yahoo.com/extension.js#L80

probably related to this other issue #164.

SpaceWolf727, 2022-12-31:
Still getting the error, extension ran fine for a day, but after reboot started throwing the error.

Tried commenting lines as advised, no luck.

vachaj11, 2023-02-13:
For me, this bug was triggered by enabling Catalyst as GPU sensor provider. 
To disable it without having access to the extension, I first:
1. Uninstalled the extension
2. Using dconf Editor turned off the option use-gpu-aticonfig (path: org/gnome/shell/extensions/freon/use-gpu-aticonfig)
3. Reinstalled the extension
Hopefully this helps!

thusvill, 2023-05-05:
> For me, this bug was triggered by enabling Catalyst as GPU sensor provider. To disable it without having access to the extension, I first:
> 
> 1. Uninstalled the extension
> 2. Using dconf Editor turned off the option use-gpu-aticonfig (path: org/gnome/shell/extensions/freon/use-gpu-aticonfig)
> 3. Reinstalled the extension
> 
> (or you can probably just log out, log in without uninstalling the extension) Hopefully this helps! edit: Also, this issue occured after I started using Wayland instead of X11, so I suspect that might have something to do with it.

Worked for me

perkelix, 2023-06-30:
Same here. What fixed it was disabling Catalyst, logging out, and logging back in.

closed 2023-12-02 by UshakovVasilii