UshakovVasilii/gnome-shell-extension-freon #244: Add support for gnome-shell 43

tjaalton, 2022-08-31:
Hi, seems that freon works fine with 43, after metadata.json is changed to support it.

dhvcc, 2022-10-21:
Fine on Fedora 37 with Gnome 43. I just disabled version check `gsettings set org.gnome.shell disable-extension-version-validation "true"` until metadata is updated and pushed.

aral, 2022-10-21:
Any word on what needs to happen for this to get pushed out? Looks like the fix was implemented about a month ago in https://github.com/UshakovVasilii/gnome-shell-extension-freon/commit/c655ccb6cfdbd132100a6cd70cda82528a1b4599?

dhvcc, 2022-10-23:
No idea mate, I guess we should ping the maintainer
@UshakovVasilii 👋

UshakovVasilii, 2022-10-23:
I send new version from https://github.com/UshakovVasilii/gnome-shell-extension-freon/tree/without-ipmi branch to https://extensions.gnome.org/ (main branch has problem with sudo https://github.com/UshakovVasilii/gnome-shell-extension-freon/pull/220#issuecomment-1079988557)