UshakovVasilii/gnome-shell-extension-freon #248: Please run sensers-detect as root

zw963, 2022-10-14:
Please check following screenshot too.

[image]

Thanks.

UshakovVasilii, 2022-10-17:
Did you execute `sensors-detect` under root? ))

zw963, 2022-10-17:
No, i guess this extension want me run sensors-detect as root, but it not.

UshakovVasilii, 2022-10-18:
Please check `sensors` command output.

zw963, 2022-10-18:
```
 ╰─ $ 1  sensors
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +65.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +59.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +58.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +59.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +65.0°C  (high = +100.0°C, crit = +100.0°C)

pch_cannonlake-virtual-0
Adapter: Virtual device
temp1:        +55.0°C  

CMB0-acpi-0
Adapter: ACPI interface
in0:           8.32 V  

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +34.0°C  

nvme-pci-3e00
Adapter: PCI adapter
Composite:    +31.9°C  (low  = -273.1°C, high = +80.8°C)
                       (crit = +81.8°C)
Sensor 1:     +31.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +35.9°C  (low  = -273.1°C, high = +65261.8°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +32.0°C  (crit = +119.0°C)
```

UshakovVasilii, 2022-10-18:
Possible sensors utility old and not support `-j` flag. lm_sensors version should be 3.5.0 or higher.. Please try `sensors -A -j`.

Problem related to the following code:
https://github.com/UshakovVasilii/gnome-shell-extension-freon/blob/master/freon@UshakovVasilii_Github.yahoo.com/extension.js#L597
https://github.com/UshakovVasilii/gnome-shell-extension-freon/blob/master/freon@UshakovVasilii_Github.yahoo.com/commandLineUtil.js#L39
https://github.com/UshakovVasilii/gnome-shell-extension-freon/blob/master/freon@UshakovVasilii_Github.yahoo.com/sensorsUtil.js#L12

zw963, 2022-10-18:
```
╰─ $ sensors -A -j
{
   "coretemp-isa-0000":{
      "Package id 0":{
         "temp1_input": 69.000,
         "temp1_max": 100.000,
         "temp1_crit": 100.000,
         "temp1_crit_alarm": 0.000
      },
      "Core 0":{
         "temp2_input": 63.000,
         "temp2_max": 100.000,
         "temp2_crit": 100.000,
         "temp2_crit_alarm": 0.000
      },
      "Core 1":{
         "temp3_input": 64.000,
         "temp3_max": 100.000,
         "temp3_crit": 100.000,
         "temp3_crit_alarm": 0.000
      },
      "Core 2":{
         "temp4_input": 69.000,
         "temp4_max": 100.000,
         "temp4_crit": 100.000,
         "temp4_crit_alarm": 0.000
      },
      "Core 3":{
         "temp5_input": 66.000,
         "temp5_max": 100.000,
         "temp5_crit": 100.000,
         "temp5_crit_alarm": 0.000
      }
   },
   "pch_cannonlake-virtual-0":{
      "temp1":{
         "temp1_input": 60.000
      }
   },
   "CMB0-acpi-0":{
      "in0":{
         "in0_input": 8.289
      }
   },
   "iwlwifi_1-virtual-0":{
      "temp1":{
         "temp1_input": 30.000
      }
   },
   "nvme-pci-3e00":{
      "Composite":{
         "temp1_input": 35.850,
         "temp1_max": 80.850,
         "temp1_min": -273.150,
         "temp1_crit": 81.850,
         "temp1_alarm": 0.000
      },
      "Sensor 1":{
         "temp2_input": 35.850,
         "temp2_max": 65261.850,
         "temp2_min": -273.150
      },
      "Sensor 2":{
         "temp3_input": 36.850,
         "temp3_max": 65261.850,
         "temp3_min": -273.150
      }
   },
   "acpitz-acpi-0":{
      "temp1":{
         "temp1_input": 32.000,
         "temp1_crit": 119.000
      }
   }
}
```

How to fix it? thanks

UshakovVasilii, 2022-10-19:
It is strange. How did you install extension? Did you use https://extensions.gnome.org ?
What Gnome version do you use?

zw963, 2022-10-19:
> Did you use https://extensions.gnome.org/ ?

Yes, need input root password when install this extension.

>  What Gnome version do you use?

I use arch linux, GNOME + Wayland. 

```sh
 ╰─ $ pacman -Q |grep gnome
gnome-autoar 0.4.3-2
gnome-backgrounds 42.0-1
gnome-bluetooth-3.0 42.4-1
gnome-books 40.0-1
gnome-boxes 42.3-1
gnome-browser-connector 42.1-3
gnome-calculator 42.2-1
gnome-calendar 42.2-1
gnome-characters 42.0-1
gnome-clocks 42.0-1
gnome-color-manager 3.36.0+r25+g4aab8b59-1
gnome-contacts 42.0-1
gnome-control-center 42.3-2
gnome-desktop 1:42.4-1
gnome-desktop-4 1:42.4-1
gnome-desktop-common 1:42.4-1
gnome-disk-utility 42.0-1
gnome-epub-thumbnailer 1.6-2
gnome-font-viewer 42.0-1
gnome-keyring 1:42.1-1
gnome-logs 42.0-1
gnome-maps 42.3-1
gnome-menus 3.36.0-2
gnome-music 1:42.1-1
gnome-online-accounts 3.44.0-1
gnome-photos 1:42.0-1
gnome-remote-desktop 42.4-1
gnome-screenshot 41.0+r25+g45f08f0-1
gnome-session 42.0-1
gnome-settings-daemon 42.2-1
gnome-shell 1:42.5-1
gnome-shell-extension-appindicator 46-1
gnome-shell-extensions 42.3-1
gnome-software 42.4-1
gnome-system-monitor 42.0-1
gnome-terminal 3.44.1-1
gnome-tweaks 42beta+r14+g2d9d76c-2
gnome-usage 3.38.1-1
gnome-user-docs 42.0-1
gnome-user-share 3.34.0-2
gnome-video-effects 0.5.0+4+g9554041-2
gnome-weather 42.0-1
libgnomekbd 1:3.28.1-1
xdg-desktop-portal-gnome 42.3-1

```

zw963, 2022-10-19:
I test on https://extensions.gnome.org/extension/1145/sensory-perception/, this extension works.

Altough, freon used to work too.

zw963, 2023-12-06:
I ran sensors-detect as root and pressing `yes` for all answer,  but this extension still not work.

zw963, 2023-12-09:
Hi, i install a fresh new linux on a new installed AMD 7840HS laptop, this extension is work well no any settings except install extension.

I don't know why this issue happen on my previous laptop, anyway, i am okay to close this issue, or, if you want, i can offer the detail of my previous laptop if any question.

closed 2023-12-09 by zw963