UshakovVasilii/gnome-shell-extension-freon #250: Feature request: Poll nvidia temperature status only if GPU is awake

andrewmackrodt, 2023-01-03:
With Optimus laptops the nvidia dGPU can be suspended to enhance battery life. `nvidia-smi` causes the GPU to wake-up, thus consuming more power. This feature request is for Freon to only poll nvidia-smi if the dGPU is already awake.

The `power_state` can be queried via `/sys/devices/../../power_state`, e.g.:

```sh
find /sys/module/nvidia/drivers/pci\:nvidia/ -mindepth 1 -maxdepth 1 -type l -name '*:*' | xargs -I@ sh -c 'echo $(basename "@") $(cat "@/power_state")'
```

When `power_state = D3cold`, temperature should not be queried.

Alternatively, disabling nvidia temperature reporting if using battery could work well as well.

maniacx, 2023-02-20:
Wow. First thing it do when i install ubuntu is install extension and freon.
Never realised it was keeping dgpu awake.
I installed supergfxctl because of this to manually switch OFF dgpu. Today yesterday I disable freon for some reason and I noticed supergfxctl was showing dgpu is suspended in hybrid mode. It is then i remembered reading your issue.
I thought my laptop didnt have optimus... :) But is does have it , and it was freon keeping it awake

retrixe, 2023-05-04:
This will probably need some heuristics, I'm thinking of submitting a PR which initially does this check using `/proc/driver/nvidia/gpus/<ID>/power` (which isn't enough on its own unless you slow down your polling), and a follow up commit which checks `nvidia-smi` running processes and slows down polling to every 30 seconds (for me, the GPU sleeps 20 seconds after running `nvidia-smi` on Wayland) if `gnome-shell`, `Xorg` or neither are running.

retrixe, 2023-05-15:
I've created a PR which fixes this without any need to reduce the poll rate :^) feel free to test it out.

Funny thing is that getting the temperature and other misc GPU info keeps the GPU alive on Windows too, I observe the same behaviour with Intel Arc Control lol