UshakovVasilii/gnome-shell-extension-freon #253: For AMD Radeon cards, you DON'T need aticonfig from the AMD Catalyst suite.

Kerrung, 2023-02-14:
Currently I have a AMD Radeon RX 550 and Arch Linux default Linux kernel. So I can see temperature by default without the AMD Catalyst suite. But your wiki talks about require the AMD Catalyst suite for that.

illwieckz, 2023-12-15:
Actually this is more a wiki issue saying something that is now wrong than a software bug.

Though at some point the aticonfig support could be removed one day. I know for sure that even in recent years some people were still running fglrx on GCN1 and GCN2 hardware because of OpenCL (see that _[OpenCL on Linux: state of AMD drivers is now worse than it was back in the days of fglrx](https://rebatir.fr/post/2022-01-25-OpenCL_on_Linux_state_of_AMD_drivers_is_now_worse_than_it_was_back_in_the_days_of_fglrx/)_ article from last year). Actually until the [Mesa rusticl driver was released this year](https://www.phoronix.com/news/Mesa-22.3-Rusticl-Merged) (and in fact it's stilll not enabled by default), fglrx was still the only viable option for OpenCL on GCN1 cards on Linux. It's hard to believe but fglrx may still be the only viable option for some OpenCL applications on GCN1 cards in 2023 and it will be still true in 2024.

illwieckz, 2023-12-15:
I updated the page to say the user should not use the _Catalyst_ option unless he is the last man standing with that driver for some specific niche need:

- https://github.com/UshakovVasilii/gnome-shell-extension-freon/wiki/Dependency

closed 2023-12-15 by illwieckz