UshakovVasilii/gnome-shell-extension-freon #254: No language source files provided

ViBE-HU, 2023-02-25:
No `.po` files. Why?

IsaacDias2020, 2023-03-05:
Same question.

UshakovVasilii, 2023-03-22:
I reverted `.po` files. Looks like I removed them, I don't know reason, I think it was mistake...
Thanks!

closed 2023-03-22 by UshakovVasilii

ViBE-HU, 2023-03-27:
Thanks. I thought that hungarian is poorly translated and that's why the plugin is still english on my system. But as I see it is a different issue. May the last compilation is made before the translation has been created?

UshakovVasilii, 2023-03-29:
I recompiled `.mo` file, there is no diff
```
msgfmt po/hu.po -o locale/hu/LC_MESSAGES/freon.mo
```

ViBE-HU, 2023-03-29:
Hm, that's pretty strange then. I'm on Ubuntu 20.04, everything is up to date and the whole application is in english somehow. The extension version is 40.