UshakovVasilii/gnome-shell-extension-freon #266: Update for G45

gilvbp, 2023-08-18:
Hi, could you update to G45 version?
This tutorial might help:
https://gjs.guide/extensions/upgrading/gnome-shell-45.html#esm
Best regards.

If you need help, I could help with some code. Here's the official link. 

PS: Here it's a link to an extension that´s already working on G45 https://github.com/ionutbortis/gnome-bedtime-mode/tree/gnome45

dinhokusanagi, 2023-09-15:
https://github.com/ionutbortis/gnome-bedtime-mode/tree/gnome45

ERRO

menocheck, 2023-09-24:
Hi, I've started a pull request with some initial work: #267

SeokYoung-Ahn, 2023-10-13:
It doesn't work on GNOME 45, I know you're busy, but please check. Please fix it. Thank you.

syahn76, 2023-10-20:
Thank you so much for the update to gnome 45.

dinhokusanagi, 2023-10-20:
thanks for the update

closed 2023-12-02 by UshakovVasilii