UshakovVasilii/gnome-shell-extension-freon #277: Freon only shows GPU temp for 1 sec while I run nvidia-smi in a terminal

eobet, 2024-01-08:
What step of installation did I mess up?

I've recently upgraded from Fedora 38 to 39, and Gnome 45 which (perhaps due to my distro) removed all extensions so I had to set them up again...

eobet, 2024-08-12:
This behavior went away for a few months, and now it's back with the latest update. 😑

retrixe, 2025-03-14:
This is intended behaviour, see #259 and [NVIDIA documentation](https://download.nvidia.com/XFree86/Linux-x86_64/570.124.04/README/dynamicpowermanagement.html)

TL;DR the GPU has been put into deep sleep by the system to conserve power and extend battery life, this includes powering off the temperature sensor in the GPU too

Running `nvidia-smi` powers the GPU back on, resulting in the behaviour you see

You can turn off runtime D3 sleep as mentioned in the NVIDIA documentation (though I can't possibly see why you would, ideally you want the GPU to be off as long as possible), otherwise this issue can be closed

eobet, 2025-03-14:
Thank you for the information.

In order to be less confusing to a new user, I would propose then that instead of reading "N/A" which reads like an error, it should instead say "zZ" which tells you about the reason behind the behavior you're experiencing.

retrixe, 2025-03-17:
Possible, but I suspect "zzz" might be just as confusing to some people and it would require some effort to implement

It's probably for the best that this issue exists, in case people are confused about it, they will most likely stumble upon this issue here

If maintainers chime in, the effort could probably be undertaken by someone