UshakovVasilii/gnome-shell-extension-freon #280: Gnome 46 support

bhundven, 2024-03-21:
It would be nice to add Gnome 46 support.

sidt4, 2024-03-22:
Fails with the following error in Gnome Shell 46.0.

```
Mar 21 18:32:56 linux gnome-shell[22033]: Extension freon@UshakovVasilii_Github.yahoo.com: TypeError: this._menuLayout.add is not a function

      Stack trace:
	_createHotItem@file:///home/sid/.local/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/extension.js:179:30
	Freon_FreonMenuButton@file:///home/sid/.local/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/extension.js:108:18
	enable@file:///home/sid/.local/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/extension.js:984:27
	_callExtensionEnable@resource:///org/gnome/shell/ui/extensionSystem.js:266:38
	loadExtension@resource:///org/gnome/shell/ui/extensionSystem.js:478:32
	async*reloadExtension@resource:///org/gnome/shell/ui/extensionSystem.js:526:20
	async*_onVersionValidationChanged@resource:///org/gnome/shell/ui/extensionSystem.js:672:24
	async*_loadExtensions/<@resource:///org/gnome/shell/ui/extensionSystem.js:740:18
	@resource:///org/gnome/shell/ui/init.js:21:20
```

sidt4, 2024-03-22:
### Gnome 46 upgrade reference:

https://gjs.guide/extensions/upgrading/gnome-shell-46.html

szpak, 2024-05-16:
It seems to be implemented by #281 and could be closed.

closed 2025-10-23 by UshakovVasilii