UshakovVasilii/gnome-shell-extension-freon #293: use nvme-cli breaks the extension

alexwbaule, 2025-04-15:
Debugging the extension, i found the error.

To get the temperature, you use the "DevicePath"  using this:
```
var smart_log = getNvmeData(`smart-log ${device["DevicePath"]}`);
```

but in the version 2.12 of the nvme, the "DevicePath" dont exists anymore.

for my device, there is:

```
{
  "Devices":[
    {
      "HostNQN":"nqn.2014-08.org.nvmexpress:uuid:e659a1a8-96d6-0000-0000-000000000000",
      "HostID":"e659a1a8-96d6-0000-0000-000000000000",
      "Subsystems":[
        {
          "Subsystem":"nvme-subsys0",
          "SubsystemNQN":"nqn.2020-04.com.kingston:nvme:nvm-subsystem-sn-50026B76863B1C8E",
          "Controllers":[
            {
              "Controller":"nvme0",
              "Cntlid":"1",
              "SerialNumber":"50026B76863B1C8E",
              "ModelNumber":"KINGSTON SKC3000S1024G",
              "Firmware":"EIFK31.6",
              "Transport":"pcie",
              "Address":"0000:09:00.0",
              "Slot":"",
              "Namespaces":[
                {
                  "NameSpace":"nvme0n1",
                  "Generic":"ng0n1",
                  "NSID":1,
                  "UsedBytes":1024209543168,
                  "MaximumLBA":2000409264,
                  "PhysicalSize":1024209543168,
                  "SectorSize":512
                }
              ],
              "Paths":[]
            }
          ],
          "Namespaces":[]
        }
      ]
    }
  ]
}

```

There is a Issue here:

https://github.com/linux-nvme/nvme-cli/issues/2749

alexwbaule, 2025-04-15:
There is a atualization, reverting back.
The version 2.13 fix that.

tulir, 2025-04-19:
Looks like the broken nvme-cli version (2.11) was shipped in Ubuntu 25.04

karypid, 2025-06-02:
I am on Bluefin (fedora-based) using `nvme-cli-2.13-1.fc42.x86_64`

My SSD give "NaN" and looking at command line I get:

```
# nvme smart-log /dev/nvme0 -o json
{
  "critical_warning":0,
  "temperature":318,
  "avail_spare":100,
  "spare_thresh":5,
  "percent_used":0,
  "endurance_grp_critical_warning_summary":0,
  "data_units_read":58060006,
  "data_units_written":53591344,
  "host_read_commands":317616342,
  "host_write_commands":989496395,
  "controller_busy_time":2057,
  "power_cycles":167,
  "power_on_hours":6676,
  "unsafe_shutdowns":74,
  "media_errors":0,
  "num_err_log_entries":0,
  "warning_temp_time":0,
  "critical_comp_time":0,
  "thm_temp1_trans_count":42,
  "thm_temp2_trans_count":0,
  "thm_temp1_total_time":566,
  "thm_temp2_total_time":0
}
```

Is this the correct command? Am I affected by the same issue (i.e. DevicePath missing?)